OpenMotorDrive/framework #24: Comply with UAVCAN standard WRT bus errors and anonymous messages

jschall, 2017-11-16:
Per [http://uavcan.org/Specification/4._CAN_bus_transport_layer/](http://uavcan.org/Specification/4._CAN_bus_transport_layer/):
> Since the discriminator is only 14 bits long, the probability of having multiple nodes that are emitting CAN frames with the same CAN ID but different data is higher than 0.006% (which is significant). Therefore, the protocol must account for possible errors on the CAN bus triggered by CAN ID collisions. In order to comply with this requirement, UAVCAN requires all nodes to immediately abort transmission of all anonymous transfers once an error on the CAN bus is detected. This measure allows to prevent the bus deadlock that may occur if the automatic retransmission on bus error is not suppressed.

jschall, 2017-11-25:
I think this may be as simple as enabling non automatic re-transmission mode during dynamic node ID allocation. Can @pavel-kirienko confirm?

pavel-kirienko, 2017-11-25:
Yes I can confirm.

jschall, 2017-12-03:
will be fixed with #49

closed 2017-12-03 by jschall