OpenMotorDrive/framework #76: bootloader is way too large for STM32F3

jschall, 2018-01-31:
Current size of the bootloader built for F4 is 17912 bytes. Need to save over 5k to fit it into our allotted 12k.

The only way that this is likely to happen is if many or all of the following changes are implemented:
- Eliminate ChibiOS/RT for the bootloader, go bare metal and single threaded
- Eliminate or dramatically lighten the pubsub and worker thread modules
- Ultra-lightweight UAVCAN stack for the bootloader
- Truncate the vector table so that unused vector table entries are used for code

pavel-kirienko, 2018-01-31:
Have you checked out the PX4 Brickproof Bootloader? It supports UAVCAN and
takes under 8K on all stm32 down to STM32F1xx. It is utilized in Sapog, for
example.

jschall, 2018-02-01:
@pavel-kirienko Thanks. IIRC, I considered that, but I wanted to deeply understand the logic by writing my own.

FYI: We also have another bootloader (https://github.com/openmotordrive/bootloader) based on libopencm3 that is far smaller. This issue refers to the port of that bootloader onto openmotordrive/framework, which apparently adds about 8k of flash usage.

The reason for porting the bootloader was that I'd like to have a common app and bootloader build system, and share as much code as possible between the app and bootloader. Additionally, more effort has gone into ensuring that the framework's UAVCAN stack complies with the UAVCAN standard (such as using NART mode when the UAVCAN node is anonymous).